nasa/XPlaneConnect #193: Question : Example how to send cmd to xpl

Andrew738, 2019-12-12:
Hi,

Sending datarefs and receiving works fine but sending just a cmd (xpl command) is still not working. I just copied the program i made. The plugin is working fine.

Maybe the datagram is not correct. If you can point me to the right direction that would be great!

Test program:

// XP Test Conn.cpp : This file contains the 'main' function. Program execution begins and ends there.
//

#include "../src/xplaneConnect.h"
#include <time.h>
#include "stdio.h"

int main(void)
{
	XPCSocket sock = openUDP("127.0.0.1");

	// Get the status of the gear handle
	// const char* dref = "sim/cockpit/switches/gear_handle_status";
	// const char* dref = "sim/cockpit/radios/nav1_freq_hz";
	// float value = 0;
	/// int size = 1;
	/*getDREF(sock, dref, &value, &size);
	printf("The value is %f.\n", value);*/

	const char* ndref = "sim/cockpit/radios/nav1_freq_hz";
	float value = 11540.000000;
	sendDREF(sock, ndref, &value, 1);

	// Initialize
	int i, j; // Iterator
	const char* drefs[100] =
	{
		"sim/aircraft/parts/acf_gear_deploy"
	};
	float* data[100]; // array for result of getDREFs
	int sizes[100];
	float DATA[4][9]; // Array for sendDATA
	
	// Setup
	for (int i = 0; i < 100; ++i)
	{
		data[i] = (float*)malloc(40 * sizeof(float));
		sizes[i] = 40;
	}
	for (i = 0; i < 4; i++)
	{
		for (j = 0; j < 9; j++)
		{
			data[i][j] = -998;
		}
	}
	DATA[0][0] = 14; // Gear
	DATA[0][1] = 1;
	DATA[0][2] = 0;

	// Execution
	sendDATA(sock, DATA, 1);
	int result = getDREFs(sock, drefs, data, 1, sizes);

	return 0;
	closeUDP(sock);
}

AHaberl, 2020-01-16:
I had a similar issue. I took the code from this PR: https://github.com/nasa/XPlaneConnect/pull/120/files
It adds the needed feature to send commands. I have no idea why it is not anymore on master...
However you need to re-build the plugin yourself.

Andrew738, 2020-01-16:
Thanks Andreas!

I build in the function myself and it works (-:

Kind regards,

Andre

xrb936, 2020-03-20:
> […](#)

Hi Andrew,

I am currently working on adding sendCOMM function to the plugin, but I am unable to compile the plugin with Visual Studio 2015. May I ask what OS and IDE were you using for compiling the plugin?

Thanks.
Robin

closed 2020-07-15 by jason-watkins