nasa/XPlaneConnect #195: New Python Programmer trouble with sendCOMM

LionheartUK, 2020-01-28:
Hi all,

Need a bit of help with the sendCOMM function. I read that the latest release client does not have this command in the xpc.py or the compiled plugin. Im using python to communicate between and arduino and X-Plane, I have managed to get datarefs in and out with no issues at all but when it comes to the sendCOMM command im hitting a wall. I see the since rc2 that the command was added but I cant see it in any of the xpc.py files or the plugin.

Does anyone have a compiled plugin that accepts sendCOMM commands and an xpc.py with the sendCOMM command in it?

Thank you.

Regards

Richard

epaton2, 2020-05-16:
I'm working on compiling a plugin with sendCOMM added.
Eric

jason-watkins, 2020-07-15:
See #197 and #219, we have re-added the missing code.

closed 2020-07-15 by jason-watkins