nasa/XPlaneConnect #197: Is sendCOMM still available to use?

xrb936, 2020-02-07:
Hi guys, based on the 1.3 RC2 release note, sendCOMM has been added to the plugin. But from the code of client, there is no method or function called sendCOMM in whatever language. I just want to check if I am good to use is by just add a sendCOMM function client, or I have to add these features in plugin, also? Thanks!

epaton2, 2020-05-16:
I notice the same problem.   In xplaneConnect.c, I do not see the code that was added for sendCOMM.  The only place I see the code is in #120.   This is weird because the notes for the merge states that it was added in RC2.   I'm going to manually merge the sendCOMM code into the latest pre-release RC5 and cross my fingers that it works.   It appears 10 files were modified, some from the client, and some from the X-Plane plugin.

vanmellebenjamin, 2020-05-29:
Got the same issue, and I  applied the modifications mentioned on this post (worked straight):
https://github.com/nasa/XPlaneConnect/issues/193

For information, it is useful to add a byte to the message, to manage the different kind of pressures : Once, Hold, Release. 

Here is my POC:

`void DataManager::Execute(const std::string& comm, uint8_t pressType)
	{
		Log::FormatLine(LOG_INFO, "DMAN", "Executing command (value:%s)", comm.c_str());

		XPLMCommandRef xcref = XPLMFindCommand(comm.c_str());
		if (!xcref)
		{
			// COMM does not exist
			Log::FormatLine(LOG_ERROR, "DMAN", "ERROR: invalid COMM %s", comm.c_str());
			return;
		}
		if (pressType == 0) {
			XPLMCommandOnce(xcref);
		}
		else if (pressType == 1) {
			XPLMCommandBegin(xcref);
		}
		else if (pressType == 2) {
			XPLMCommandEnd(xcref);
		}

	}`

sanderdatema, 2020-06-23:
I'm confused, where is the code if there clearly is a merge? And can someone merge it into master? After that I could compile it for the Mac.

epaton2, 2020-06-24:
I just merged it on my local machine but did not merge it on the repo. So
you won’t see it there.    I Don’t know how to do pull requests and
comparing branches and all that so I just kept it on my local machine.

vanmellebenjamin, 2020-06-24:
I did the same, I pulled and compiled my own version of the code, to have the feature available for my use case.

sanderdatema, 2020-06-27:
I added the code to the `develop` branch again in #217.

closed 2020-07-15 by jason-watkins