nasa/XPlaneConnect #198: SendDREF fails in Python when sending a list

MadisonTBlake, 2020-02-17:
In Python the method SendDREF(s) fails when packing a list passing at the following location:

```python
if hasattr(value, "__len__"):
    if len(value) > 255:
        raise ValueError("value must have less than 256 items.")
    fmt = "<B{0:d}sB{1:d}f".format(len(dref), len(value))

    # The next line will fail
    buffer += struct.pack(fmt.encode(), len(dref), dref.encode(), len(value), value)
else:
    fmt = "<B{0:d}sBf".format(len(dref))
    buffer += struct.pack(fmt.encode(), len(dref), dref.encode(), 1, value)
```

The failure depends on the size of the array being passed. For example if passing an array of size one the following failure is reported:
`self._client.sendDREF("sim/multiplayer/controls/yoke_roll_ratio", [1])`
```
buffer += struct.pack(fmt.encode(), len(dref), dref.encode(), len(value), value)
struct.error: required argument is not a float
```

If passing an array larger than one the error is different:  
`self._client.sendDREF("sim/multiplayer/controls/yoke_roll_ratio", [0, 1])`
```
buffer += struct.pack(fmt.encode(), len(dref), dref.encode(), len(value), value)
struct.error: pack expected 5 items for packing (got 4)
```

For the DREF in question the docs state that I can accept an array of size 20:
`yoke_roll_ratio | float[20] | 860+ | yes | [-1..1] | The deflection of the axis controlling roll.`

For my purposes I did a quick and dirty fix that only works for lists up to size 2:
```python
if len(value) == 1:
    buffer += struct.pack(fmt.encode(), len(dref), dref.encode(), 1, value[0])
elif len(value) == 2:
    buffer += struct.pack(fmt.encode(), len(dref), dref.encode(), 2, value[0], value[1])
else:
    raise "Array of size " + str(len(value)) + " Not implemented"
```

After putting this kludge in the DREFS in question adjust the simulator as they are supposed to.

jamesdunham, 2020-05-09:
Thanks for writing up your solution. I was running into this periodically too, and your approach worked for me.

I had longer arrays, so I did:

```
    buffer += struct.pack(fmt.encode(), len(dref), dref.encode(), 2, *value)
```

suryasekhar06jemsbond-lab, 2025-10-04:
1. Python3/src/xplaneconnect/client.py — full client with _build_dref_packet helper and sendDREF that correctly expands list/tuple values into struct.pack.

2. Python3/tests/test_senddref.py — two tests: single-value and multi-value packing checks (no network I/O).

---
# -*- coding: utf-8 -*-
"""
xplaneconnect.client
Drop-in client module for XPlaneConnect with correct packing for SendDREF values.
Provides:
- _build_dref_packet(dref, value) -> bytes : returns the binary packet for a DREF message
- XPlaneConnect class with sendDREF(dref, value) which sends via UDP
"""
from __future__ import annotations

import socket
import struct
from typing import Iterable, Union, Tuple

__all__ = ["XPlaneConnect", "_build_dref_packet"]

DEFAULT_ADDR = ("127.0.0.1", 49009)  # local default; adjust per repo convention

def _is_sequence(v) -> bool:
    return isinstance(v, (list, tuple))

def _build_dref_packet(dref: str, value: Union[float, int, Iterable[Union[float, int]]]) -> bytes:
    """
    Build a DREF packet for X-Plane.

    The packet layout (this helper's convention):
    - 4 bytes ASCII header 'DREF'
    - 1 byte: length of dref (L)
    - L bytes: dref name encoded as UTF-8 (exactly L bytes)
    - 1 byte: number of float values (N)  (for scalar N==1)
    - N floats (little-endian, 32-bit)

    Notes:
    - Raises ValueError if sequence length > 255 or dref length > 255.
    - Ensures each numeric value is cast to float.
    """
    if not isinstance(dref, str):
        raise TypeError("dref must be a string")

    dref_bytes = dref.encode("utf-8")
    len_dref = len(dref_bytes)
    if len_dref > 255:
        raise ValueError("dref length must be <= 255 bytes")

    # Header
    header = b"DREF"

    # If scalar, normalize to single-element tuple
    if _is_sequence(value):
        vals_tuple = tuple(float(x) for x in value)
    else:
        vals_tuple = (float(value),)

    n_vals = len(vals_tuple)
    if n_vals > 255:
        raise ValueError("value must have less than 256 items.")

    # Format: <B{len_dref}sB{n_vals}f
    # We'll pack with struct.pack(fmt, len_dref, dref_bytes, n_vals, *vals_tuple)
    fmt = "<B{}sB{}f".format(len_dref, n_vals)
    try:
        payload = struct.pack(fmt, len_dref, dref_bytes, n_vals, *vals_tuple)
    except struct.error as e:
        # Re-raise with clearer message for maintainers
        raise struct.error(f"failed packing DREF payload: {e}; fmt={fmt}, len(vals)={n_vals}") from e

    return header + payload

class XPlaneConnect:
    """
    Minimal UDP client for X-Plane messages.
    Usage (example):
        c = XPlaneConnect(address=("127.0.0.1", 49009))
        c.sendDREF("sim/multiplayer/controls/yoke_roll_ratio", [1.0])
    """

    def __init__(self, address: Tuple[str, int] = DEFAULT_ADDR, timeout: float = 1.0):
        self.address = address
        self.timeout = timeout
        self._sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
        self._sock.settimeout(self.timeout)

    def close(self):
        try:
            self._sock.close()
        except Exception:
            pass

    def sendDREF(self, dref: str, value: Union[float, int, Iterable[Union[float, int]]]) -> None:
        """
        Send a DREF (dataref) set command to X-Plane.

        - dref: string dataref name
        - value: float/int or sequence of floats/ints

        This uses _build_dref_packet() to construct the packet.
        """
        packet = _build_dref_packet(dref, value)
        # Fire-and-forget UDP send
        self._sock.sendto(packet, self.address)

    # context manager support
    def __enter__(self):
        return self

    def __exit__(self, exc_type, exc, tb):
        self.close()

And

# -*- coding: utf-8 -*-
"""
Unit tests for DREF packet packing.
These tests do NOT send UDP; they validate the bytes returned by _build_dref_packet.
"""

import struct

import pytest

from xplaneconnect.client import _build_dref_packet

def _unpack_payload(packet: bytes):
    """
    Return tuple (dref_str, values_list) by parsing the helper payload format:
    packet = b"DREF" + struct.pack("<B{Ld}sB{N}f", len_dref, dref_bytes, N, *values)
    """
    assert packet.startswith(b"DREF"), "packet must start with DREF"
    payload = packet[4:]
    if len(payload) < 2:
        raise AssertionError("payload too short")
    len_dref = payload[0]
    start = 1
    dref_bytes = payload[start : start + len_dref]
    dref = dref_bytes.decode("utf-8")
    start = start + len_dref
    n_vals = payload[start]
    start = start + 1
    floats_offset = start
    floats_size = 4 * n_vals
    floats_bytes = payload[floats_offset : floats_offset + floats_size]
    if n_vals:
        vals = list(struct.unpack("<{}f".format(n_vals), floats_bytes))
    else:
        vals = []
    return dref, vals

def test_senddref_packing_single():
    dref = "sim/multiplayer/controls/yoke_roll_ratio"
    values = 1  # scalar
    packet = _build_dref_packet(dref, values)
    got_dref, vals = _unpack_payload(packet)
    assert got_dref == dref
    assert len(vals) == 1
    assert pytest.approx(vals[0], rel=1e-6) == float(values)

def test_senddref_packing_multi():
    dref = "sim/multiplayer/controls/yoke_roll_ratio"
    values = [0.0, 1.5, -2.25]
    packet = _build_dref_packet(dref, values)
    got_dref, vals = _unpack_payload(packet)
    assert got_dref == dref
    assert len(vals) == len(values)
    for a, b in zip(vals, values):
        assert pytest.approx(a, rel=1e-6) == float(b)

Run:

# from repo root
python3 -m venv .venv
. .venv/bin/activate
pip install -U pip pytest
# Make sure package import path includes Python3/src or adjust PYTHONPATH
PYTHONPATH=Python3/src pytest -q Python3/tests/test_senddref.py