nasa/XPlaneConnect #199: X-plane crashes executing sendPOSI with python library

TonyKenny666, 2020-03-14:
Hi,
I'm relatively new to X-plane and python so please excuse any omissions in my report.

Setup: 
Windows 10
X-plane 11.41
XPlaneConnect 1.3 RC5
Python 3.8.2
Running inside Visual Studio Code  

I've changed the example to python3 syntax which was mostly just adding brackets to print statements.
Executing basicExample.py causes an immediate freeze of the simulator as soon as the first sendPOSI line is executed. When it freezes, the ATC audio continues but there's also a loud sound similar to landing gear being raised/lowered, but that command hadn't executed and I also tried with that command commented out just to be sure.

Changing the position as below works as expected. 

`posi = [52.6, -1.1, 2500, 0,    0,   0,  1]`

Could this be that I don't have scenery for the location in the example and X-Plane cannot cope with it? If so, then is this a bug with X-Plane rather than with this library and maybe just a quick note might be helpful in the example to help others in future? 

XPCLog attached

[XPCLog.txt](https://github.com/nasa/XPlaneConnect/files/4332691/XPCLog.txt)

jason-watkins, 2020-03-16:
Thanks for the detailed report (and especially for including the log file). Believe it or not, this is the first time I've ever received a report of X-Plane Connect crashing/locking up X-Plane. Your guess that it's a result of not having scenery installed is as good as any guess I could make. The log file certainly doesn't show any signs of the plugin encountering any problems. Unfortunately I don't have time to set everything up to try to reproduce the issue. I'll leave this open for a bit just in case anyone else happens to have seen anything similar before and cares to comment.

I'm not aware of anything in the plugin API that would let me detect this and produce a useful message somehow, so I think the best I can do is add a note to the example pointing out this possibility.

Mammux, 2020-04-14:
More or less the same thing happened to me, but it unfroze after a couple of minutes when the scenery and texture were finished loading. I thought maybe the scenery loading changed in a recent version of x-plane, I'm actually running the 11.50b3 beta with Vulkan. The noise is probably just the stall warning of the plane.

closed 2020-07-15 by jason-watkins