nasa/XPlaneConnect #200: How to Communicate XPlane with C-code using UDP

RakeshVangaveti, 2020-04-14:
hi,
im new to xplane connect/xplane , i  want to communicate the xplane with C-code using UDP , i gone through the documentation follow the link 
https://github.com/nasa/XPlaneConnect/wiki
but i did'nt understand what they mention , i think they have implemented their own protocol for xplane , the thing is how do we implement that ,if you have any implemented program in C for XPlane UDP communication please attach here.
i hope i'll get a quick reply.
thanks,
Rakesh.

jason-watkins, 2020-04-14:
This project does indeed implement its own protocol that replaces and expands on the communication capabilities built in to X-Plane. This issue tracker is not an appropriate venue for questions about built in X-Plane functionality.

closed 2020-04-14 by jason-watkins

RakeshVangaveti, 2020-04-14:
sorry about that, tell me one thing is there any possibility to connect x plane by c-code using udp

teubert, 2020-04-14:
Hello @RakeshVangaveti you can find some documentation for the XPlane UDP protocol here: http://www.nuclearprojects.com/xplane/xplaneref.html

As @jason-watkins mentions, you can also use the XPC plugin and clients (this project) to communicate over UDP with XPlane using our own similar but extended protocol

RakeshVangaveti, 2020-04-18:
> Hello @RakeshVangaveti you can find some documentation for the XPlane UDP protocol here: http://www.nuclearprojects.com/xplane/xplaneref.html
>
> As @jason-watkins mentions, you can also use the XPC [...]

thanks @teubert  FYI & link

RakeshVangaveti, 2020-04-27:
hello @jason-watkins @teubert , 

As @teubert mentions "you can also use the XPC plugin and clients (this project) to communicate over UDP with XPlane using our own similar but extended protocol"

i found TestScripts directory in XPlaneConnect (having c,java,matlab,python) i think those are sample codes , how do we use them to communicate xplane ? 
there is a C testscripts inside testscript's directrory, want to communicate xplane with C test scripts how do we do them?
pardon me if i ask wrong..!
thanks in advance...
please give a reply.