nasa/XPlaneConnect #201: Actual throttle locked after sendCTRL

Mammux, 2020-04-14:
I created a simple python program that basically has only one call:

sendCTRL(0.0, 0.0, 0.0, 0.8)

It works as expected. However, after this call, moving my Saitek throttle (or pressing F1/F2) moves the representation of the physical throttle lever in the cockpit, but it doesn't change the throttle setting of the aircraft (or the RPM). The other flight controls are still controllable from x-plane. If I turn on the data output of throttle commanded and throttle actual, the throttle commanded changes as I move my throttle lever, while throttle actual is now fixed at 0.8. If I reload the airplane, I can control the throttle with the lever again.

This is in the newest X-Plane beta, 11.50b3

jason-watkins, 2020-04-15:
I suspect that's probably because of [this](https://github.com/nasa/XPlaneConnect/blob/master/xpcPlugin/MessageHandlers.cpp#L262) line. Unfortunately, looking back at the git history, it appears that line dates to before we open-sourced the software and I no longer remember (if I ever knew) why it's necessary.

You could try removing that line and recompiling the plugin. If everything works as expected without setting the override, perhaps we could remove it by default.

Mammux, 2020-04-15:
Ah, that explains it. Probably the line is there because otherwise the sim would just pick up the active throttle setting from a connected lever again.

closed 2020-04-15 by Mammux

sorrow550, 2020-09-13:
Hello, I encountered the same problem as you, namely actual throttle locked after sendCTRL. How did you solve this problem?

Mammux, 2020-09-13:
I didn't, I avoided it. The way to solve it is probably dependent on what you seek to accomplish.