nasa/XPlaneConnect #212: Send data in real time through sendposi(), resulting in model aircraft chattering

shanccshuirr, 2020-06-04:
My data is obtained every 15ms. I try to send this data to xplane11 model every 15ms, using sendposi(), but the aircraft is shaking. According to experience, because the model's own flight is not disconnected, the data I send makes the aircraft switch between the two modes. Please ask me how I solve this problem.

jason-watkins, 2020-06-04:
15ms is *extremely* fast. Even if you're running at a steady 60fps, that's slightly more than one update per frame. If you're sending updates that fast anyway, why don't you disable the flight model?

Unfortunately this just isn't something that X-Plane is really set up to do. `sendPOSI` is really intended to be used either with the physics model turned off or as a one-off command to move the aircraft to a starting position. If you really need to send fast updates with the flight model enabled, I'd suggest that you need to drop the update rate at least a little bit, but more importantly you're going to have to dig in and set basically every DREF related to the flight model (i.e. rates, control surfaces, etc) in addition to just position and orientation.

shanccshuirr, 2020-06-04:
Thank you for your answer. Please tell me how to disable the flight model. I tried, but I didn't find a way?

jason-watkins, 2020-06-04:
The [`pauseSim`](https://github.com/nasa/XPlaneConnect/wiki/pauseSim) command will pause the physics engine.

shanccshuirr, 2020-06-04:
Thank you very much for your help. I have achieved my goal according to your method

closed 2020-06-04 by shanccshuirr