nasa/XPlaneConnect #214: Is it possible to send one part of an array?

sanderdatema, 2020-06-18:
Quite a few datarefs are arrays. You can only request the whole array (or can you?) and you can only send back the whole array. So if you want to change only one item, you need to get the current array first, change one item and send it back.

Is there another way?

jason-watkins, 2020-06-18:
You are correct about the current plugin functionality. The X-Plane SDK does support getting and setting parts of the array (e.g. [XPLMGetDatavf](http://www.xsquawkbox.net/xpsdk/mediawiki/XPLMGetDatavf), so it should be fairly easy to add new commands that take an offset and count. Pull requests are always welcome!

sanderdatema, 2020-06-19:
Thanks, but I'm afraid I lack the knowledge to be of any help...

closed 2020-06-19 by jason-watkins