nasa/XPlaneConnect #215: How to distribute binaries (if allowed to do so)

sanderdatema, 2020-06-26:
I'm working on a script that will need to be able to send commands. For that I need to compile the XplaneConnect plugin myself, since the `master` does not contain that function, but another pull request (#120) does.

If I were to publish my script on the X-Plane forums. Would I be allowed to attach that custom binary?

jason-watkins, 2020-06-26:
I really have no idea how on Earth that code disappeared from the mainline branches. It's on my todo list to look into it at some point, but if someeone just re-submitted the code in a new PR I would almost certainly get around to it sooner.

In the mean time, I don't think it's appropriate to distribute a custom binary with the X-Plane Connect name attached to it.

sanderdatema, 2020-06-26:
Done: #216

The build seems to fail for MacOS: `xcodebuild: error: SDK "macosx10.14" cannot be located.`. I suppose that needs to be fixed on your side? The others (Win/Linux) are fine.

closed 2020-06-27 by sanderdatema