nasa/XPlaneConnect #221: XPlane sendctrl not working

vegaf1, 2020-07-15:
Hello, 
I am using XPlane 11 and the XPlaneConnect-1.2.1 plugin. 
When I try the basicExample.py, the sendctrl function does not seem to work (the plane just faces down and crashes). Is there a dataref that I am supposed to trigger in order for this function to work? I am trying to control both the player aircraft and an non player aircraft by sendctrl. Is there any way I can fix this? 

Thanks.

jason-watkins, 2020-07-15:
There's not really much to go on in your description... You could try posting the XPC log here and I can take a look to make sure the command is being processed correctly by the plugin, but I rather expect that it is. The physics simulation is very picky, so the exact state of the sim when the command is processed and the control inputs provided can cause all sorts of weird results.

vegaf1, 2020-07-15:
This is the XPC Log: 

X-Plane Connect [Version 1.2.1]
Compiled May  3 2017 10:51:43
Copyright (c) 2013-2017 United States Government as represented by the
Administrator of the National Aeronautics and Space Administration.
All Rights Reserved.

This file contains debugging information about the X-Plane Connect plugin.
If you have technical issues with the plugin, please report them by opening
an issue on GitHub (https://github.com/nasa/XPlaneConnect/issues) or by
emailing Christopher Teubert (christopher.a.teubert@nasa.gov).

X-Plane Version: 11410
Plugin Manager Version: 301
Host Application ID: 1
Log file generated on 2020-07-15.
[14:58:35]  INFO|EXEC|Plugin Start
[14:58:35] TRACE|DMAN|Initializing drefs
[14:58:35] TRACE|SOCK|Opening socket (port:49009)
[14:58:35] TRACE|MSGH|Setting socket
[14:58:35]  INFO|EXEC|Plugin Enabled, sockets opened
[14:58:35]  INFO|EXEC|Debug Logging Enabled (Verbosity: 6)
[15:00:08] TRACE|MESG|Read message with length 26
[15:00:08] TRACE|MSGH|Initializing handlers
[15:00:08]  INFO|MSGH|Handling message from 127.0.0.1:49044
[15:00:08] DEBUG|MSGH|New connection. ID=0, Remote=127.0.0.1:49044
[15:00:08] TRACE|DBUG| 47 45 54 44 00 01 13 73 69 6d 2f 74 65 73 74 2f 74 65 73 74 5f 66 6c 6f 61 74
[15:00:08] DEBUG|DBUG|Head: GETD Size: 26
[15:00:08] DEBUG|DBUG|    #1/1 (size:19) sim/test/test_float
[15:00:08] TRACE|GETD|DATA Requested: New Request for connection 0 (1 data refs)
[15:00:08] TRACE|DMAN|Entered Get(string, float*, int)
[15:00:08]  INFO|DMAN|Get DREF sim/test/test_float (x:98AF878) Type: 2
[15:00:08]  INFO|DMAN| -- value was 0.000000
[15:00:08]  INFO|SOCK|Send succeeded. (remote: 127.0.0.1:49044)
[15:00:08] TRACE|MESG|Read message with length 34
[15:00:08]  INFO|MSGH|Handling message from 127.0.0.1:49044
[15:00:08] DEBUG|MSGH|Existing connection. ID=0, Remote=127.0.0.1:49044
[15:00:08] TRACE|DBUG| 50 4f 53 49 00 00 93 18 16 42 53 23 f4 c2 00 40 1c 45 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 3f
[15:00:08] DEBUG|DBUG|Head: POSI Size: 34 AC:0 Pos:(37.524 -122.069 2500) Orient:(0 37.524 -122.069) Gear:1
[15:00:08] TRACE|POSI|Message Received (Conn 0)
[15:00:08]  INFO|DMAN|Setting position (37.523998, -122.068993, 2500.000000) for aircraft 0
[15:00:08]  INFO|DMAN|Get DREF 1700 (x:987AE70) result 0.427511 for a/c 0
[15:00:08]  INFO|DMAN|Get DREF 1701 (x:987AEF8) result 0.229873 for a/c 0
[15:00:08]  INFO|DMAN|Get DREF 1702 (x:987AF80) result 119.964340 for a/c 0
[15:00:08]  INFO|DMAN|Setting DREF 2100 (x:98003A0) to -6093.565225 for a/c 0
[15:00:08]  INFO|DMAN|Setting DREF 2101 (x:9800428) to 2496.529671 for a/c 0
[15:00:08]  INFO|DMAN|Setting DREF 2102 (x:98004B0) to -2674.757944 for a/c 0
[15:00:08]  INFO|DMAN|Setting DREF 2000 (x:9800538) to 37.523998 for a/c 0
[15:00:08]  INFO|DMAN|Setting DREF 2001 (x:98005C0) to -122.068993 for a/c 0
[15:00:08]  INFO|DMAN|Setting DREF 2002 (x:9800648) to 2500.000000 for a/c 0
[15:00:08]  INFO|DMAN|Setting orientation (0.427511, 0.229873, 119.964340) for aircraft 0
[15:00:08]  INFO|DMAN|Setting DREF 1700 (x:987AE70) to 0.427511 for a/c 0
[15:00:08]  INFO|DMAN|Setting DREF 1701 (x:987AEF8) to 0.229873 for a/c 0
[15:00:08]  INFO|DMAN|Setting DREF 1702 (x:987AF80) to 119.964340 for a/c 0
[15:00:08]  INFO|DMAN|Setting DREF 1704 (x:987D120) (4 values) for a/c 0
[15:00:08]  INFO|DMAN|Setting orientation (0.000000, 0.000000, 0.000000) for aircraft 0
[15:00:08]  INFO|DMAN|Setting DREF 1700 (x:987AE70) to 0.000000 for a/c 0
[15:00:08]  INFO|DMAN|Setting DREF 1701 (x:987AEF8) to 0.000000 for a/c 0
[15:00:08]  INFO|DMAN|Setting DREF 1702 (x:987AF80) to 0.000000 for a/c 0
[15:00:08]  INFO|DMAN|Setting DREF 1704 (x:987D120) (4 values) for a/c 0
[15:00:08]  INFO|DMAN|Setting gear (value:1.000000, immediate:1) for aircraft 0
[15:00:08]  INFO|DMAN|Setting DREF 1401 (x:9848758) to 1 for a/c 0
[15:00:08]  INFO|DMAN|Setting DREF 1400 (x:98235E0) (10 values) for a/c 0
[15:00:08] TRACE|MESG|Read message with length 34
[15:00:08]  INFO|MSGH|Handling message from 127.0.0.1:49044
[15:00:08] DEBUG|MSGH|Existing connection. ID=0, Remote=127.0.0.1:49044
[15:00:08] TRACE|DBUG| 50 4f 53 49 00 01 3e 19 16 42 53 23 f4 c2 00 40 1c 45 00 00 00 00 00 00 a0 41 00 00 00 00 00 00 80 3f
[15:00:08] DEBUG|DBUG|Head: POSI Size: 34 AC:1 Pos:(37.5247 -122.069 2500) Orient:(0 37.5247 -122.069) Gear:1
[15:00:08] TRACE|POSI|Message Received (Conn 0)
[... 91 lines cut ...]
[15:00:10] TRACE|SIMU|Message Received (Conn 0)
[15:00:10]  INFO|DMAN|Setting DREF 1 (x:98AAAD8) (20 values) for a/c 0
[15:00:10]  INFO|SIMU|Simulation resumed
[15:00:10] TRACE|MESG|Read message with length 50
[15:00:10]  INFO|MSGH|Handling message from 127.0.0.1:49044
[15:00:10] DEBUG|MSGH|Existing connection. ID=0, Remote=127.0.0.1:49044
[15:00:10] TRACE|DBUG| 44 52 45 46 00 27 73 69 6d 2f 63 6f 63 6b 70 69 74 2f 73 77 69 74 63 68 65 73 2f 67 65 61 72 5f 68 61 6e 64 6c 65 5f 73 74 61 74 75 73 01 00 00 00 00
[15:00:10] DEBUG|DBUG|Head: DREF Size: 50
[15:00:10] DEBUG|DBUG|    DREF (size 39) = sim/cockpit/switches/gear_handle_status
[15:00:10] DEBUG|DBUG|    Values(size 1) = 1.78017e-15
[15:00:10] TRACE|DREF|Request to set DREF value received (Conn 0)
[15:00:10]  INFO|DMAN|Setting DREF sim/cockpit/switches/gear_handle_status (x:9848758) Type: 1
[15:00:10]  INFO|DMAN| -- value was 0
[15:00:10] DEBUG|DREF|Set 1 values for sim/cockpit/switches/gear_handle_status
[15:00:14] TRACE|MESG|Read message with length 46
[15:00:14]  INFO|MSGH|Handling message from 127.0.0.1:49044
[... cut ...]
[15:00:14] TRACE|DBUG| 47 45 54 44 00 01 27 73 69 6d 2f 63 6f 63 6b 70 69 74 2f 73 77 69 74 63 68 65 73 2f 67 65 61 72 5f 68 61 6e 64 6c 65 5f 73 74 61 74 75 73
[15:00:14] DEBUG|DBUG|Head: GETD Size: 46
[15:00:14] DEBUG|DBUG|    #1/1 (size:39) sim/cockpit/switches/gear_handle_status
[15:00:14] TRACE|GETD|DATA Requested: New Request for connection 0 (1 data refs)
[15:00:14] TRACE|DMAN|Entered Get(string, float*, int)
[15:00:14]  INFO|DMAN|Get DREF sim/cockpit/switches/gear_handle_status (x:9848758) Type: 1
[15:00:14]  INFO|DMAN| -- Real value was 0, cast to 0.000000
[15:00:14]  INFO|SOCK|Send succeeded. (remote: 127.0.0.1:49044)

vegaf1, 2020-07-15:
I also flew the aircraft with the 'AI flies your airfcraft' setting under the Flight tab and I printed out the output of getctrl. I got the following:  ctrl = [1.3083007388559054e-06, -2.6579946279525757e-05, 5.7890447351383045e-06, 1, 0, 0, 0]. I then tried passing in this array to sendctrl, yet it does the same thing (heads down and crashes).

jason-watkins, 2021-06-29:
Closing due to inability to reproduce.

closed 2021-06-29 by jason-watkins