nasa/XPlaneConnect #222: Object Oriented API

frizman21, 2020-07-17:
Is this project open to or already have an Object Oriented interface to this capability?

For example, have a plane object that is receiving updates from x-plane, and one can retrieve current state of (for example) airspeed via accessor functions.

Or an autopilot object with operations such as enable/disable autopilot, setting altitude, etc. 

Less "send a message" kind of interface and a more natural experience.

I would intend to build it for Java, the language of my project. If such an API already existed in one of the other support languages, I would replicate instead of designing the API myself. 

Thoughts?

jason-watkins, 2020-07-18:
I have no objection at all to adding such and API, but my time is limited enough as it is, so I don't have much interest in supporting it. If you submit a PR, I'll review it and probably accept it, but expect me to ping you if anyone opens issues against the OO API, and expect it to be removed if it goes unsupported.

frizman21, 2020-07-18:
Understood.

The existing interface is far more exhaustive than I need for my project. I would need the portion of the API that I'm on the hook for to be scoped. For now, that would primary cockpit controls (stick, throttle, autopilot, gear,  flaps, weapons) and the aircraft state that it affects, namely position/orientation, velocity and state of everything it controls. (i.e. flaps, gear) If you have suggestions on additional aspects that should naturally also be included, please know that those suggestions are appreciated.

jason-watkins, 2020-07-18:
It doesn't need to be "complete" for me to accept it. As long as what it does is useful that's good enough for me.

jason-watkins, 2021-06-29:
Still happy to consider a PR related to this, but I don't think the issue needs to stay open.

closed 2021-06-29 by jason-watkins