nasa/XPlaneConnect #223: Publish Java library to maven repo

frizman21, 2020-07-18:
I'd like to propose, like most other open source capabilities, that the re-usable library components of the Java bindings be pushing to maven's central repository.

jason-watkins, 2020-07-18:
I know absolutely nothing about how to go about doing that. If it's something that can be automated such that the maven repo gets updated when a GitHub release is created, I have no objection to doing it. That said, I don't know if NASA already has a presence there. If they don't, there could be some significant bureaucratic hurdles to getting one set up.

frizman21, 2020-07-18:
I've always manually published my released libraries, with a Jenkins server publishing "SNAPSHOT" versions on every commit. I can do some research into Github Actions and see if this pipeline is well established. 

NASA does have a [presence](https://search.maven.org/search?q=g:gov.nasa), and another [project](https://github.com/NASA-PDS) published a recent as March. If we live in that namespace, we'll need to transfer signing artifacts or get another key wired up. Do you know the guys maintaining that project?

The most significant aspect of the work would be converting the project into the [maven folder structure](https://maven.apache.org/guides/introduction/introduction-to-the-standard-directory-layout.html).

jason-watkins, 2020-07-18:
Sure. I should be able to figure out who to talk to about getting the necessary permissions to push stuff to the gov.nasa group. It also probably doesn't absolutely have to be an automated process as long as it's easy enough that I don't feel tempted to skip it when I do a release here.

frizman21, 2020-07-30:
@jason-watkins, I've PR'd my maven folder structure re-org. See mentioned PR.

Any word from the folks at the other gov.nasa project?

jason-watkins, 2020-07-30:
Will take a look at the PR sometime this weekend (or maybe sooner, but remind me if there's no movement by Sunday afternoon)

No word yet because I completely forgot to ask. Put some feelers out now though.

jason-watkins, 2021-06-29:
Unfortunately this seems like something we don't have a process for and figuring out how to get a process in place is beyond the scope of the time I'm able to devote to this project.

PR #227 (2020-08-02): Maven Folder Structure Re-org
fixes #223
closed 2021-06-29 by jason-watkins