nasa/XPlaneConnect #225: General question about debugging the xpcPlugin with Visual Studio

epaton2, 2020-07-23:
Previously, I have only worked with the C code on the client side.  Now, I have a need to modify the xpcPlugin project in order to add a new feature. I would like to add the capability of loading a situation file from the client.   The first rewrite is complete, compiled the .xpl file, and loaded it in the plugins folder in X-Plane.  How do I debug the xpcPlugin code?  I would like to start the debug within the client code, then step into the xpcPlugin project with the sendUDP function.  I guess this is a general question because all developers who write code for the X-Plane plugin will need to know this strategy,  Thank you in advance.  This community has been very helpful to my project.  Have a great day!

jason-watkins, 2020-07-24:
Generally speaking, I debug the plugin the same way as any other code. Mostly, I try to write good minimal code that doesn't have any bugs :)

When I do need a debugger, I seem to recall that I didn't have any trouble starting up X-Plane manually and then attaching Visual Studio to the process. I don't remember any gotchas with that process, but it has been a while. If you can't get this to work, I can fire up X-Plane and see if I can get it to work and give more detailed instructions.

epaton2, 2020-07-25:
Attaching the X-Plane process was the key to getting this to work.   From the xpcPlugin project, I attached the running X-Plane process.  Then, from the C client project, I used the debugger to walk through the code, then stepped successfully into the xpcPlugin project code.   It appears that I don't have to compile the xpcPlugin dll and copy it into the plugins folder.  When you attach the process, it does this for you.  I am shocked how all of this works.  So many pieces, but they all seem to work seamlessly together.  I need more time to confirm that this is the correct method..   I'll report back later to this thread.  Thank you Jason.

closed 2021-06-29 by jason-watkins