nasa/XPlaneConnect #226: X-Plane crashes when retrieving DREFS

sanderdatema, 2020-07-24:
Using Python I retrieved a list of DREFS, I think it might be the length of the request that is causing trouble, but I'm not sure. And `Log.txt` has no info. No other plugins or custom scenery was loaded.

This is the request using `getDREFS` and `sendUDP`:
```
sendUDP b'GETD\x00"\x18AirbusFBW/OHPLightsATA21\x18AirbusFBW/OHPLightsATA24\x18AirbusFBW/OHPLightsATA26\x18AirbusFBW/OHPLightsATA27\x18AirbusFBW/OHPLightsATA28\x18AirbusFBW/OHPLightsATA29\x18AirbusFBW/OHPLightsATA30\x18AirbusFBW/OHPLightsATA31\x18AirbusFBW/OHPLightsATA34\x18AirbusFBW/OHPLightsATA35\x18AirbusFBW/OHPLightsATA49\x18AirbusFBW/OHPLightsATA70\x12AirbusFBW/BatVolts\x18AirbusFBW/OHPLightsATA32\x14AirbusFBW/ACP1Lights\x1aAirbusFBW/ADIRUSwitchArray\x1aAirbusFBW/OHPLightSwitches\x13AirbusFBW/AnnunMode\x1aAirbusFBW/PaxDoorModeArray\x16AirbusFBW/FuelOHPArray\x13AirbusFBW/APUMaster\x14AirbusFBW/APUStarter\x18AirbusFBW/APUBleedSwitch\x16AirbusFBW/ElecOHPArray\x1bAirbusFBW/EmerCallOHPButton\x18AirbusFBW/FireExOHPArray\x15AirbusFBW/HydOHPArray\x18AirbusFBW/ADRSwitchArray\x1cAirbusFBW/ENGFireSwitchArray\x18AirbusFBW/FCCSwitchArray\x19AirbusFBW/GPWSSwitchArray\x19AirbusFBW/ProbeHeatSwitch\x1cAirbusFBW/OHPBrightnessLevel\x16AirbusFBW/DUBrightness'
```

sanderdatema, 2020-07-24:
I just noticed the `"` in that string. That shouldn't be there. Gonna find out how it got there. It's not there for smaller requests.

daemotron, 2021-12-15:
I confirm GETD can cause a CTD of the simulator due to a segmentation fault when the data refs queried hold all together more data than what fits into the response buffer. The related piece of code is here:

https://github.com/nasa/XPlaneConnect/blob/master/xpcPlugin/MessageHandlers.cpp#L536

XPC uses a fixed buffer size of 4096 bytes to assemble the response. However, there is no check in place whether adding the next data ref's content will exceed the buffer size or not.

patricksurry, 2022-01-02:
:+1: also noticed while trying to cycle through all the datarefs listed in `Resources/plugins/DataRefs.txt`.    Usually will crash to desktop on the next attempt to retrieve datarefs, and shows a udp.recv(16384) error on the python client.   I initially thought it was related to the [255 byte limit per data-ref](https://github.com/nasa/XPlaneConnect/blob/81a1447d2a3d8de1c30fde4e471a7b40cbf361d6/xpcPlugin/MessageHandlers.h#L99) which is often exceed by large float or string arrays.  In those cases `XPCLog.txt` log will contain [Warning: dref size is larger than available space](https://github.com/nasa/XPlaneConnect/blob/master/xpcPlugin/DataManager.cpp#L337).   Seems like these limits could usefully be #define'd with increased default size?