nasa/XPlaneConnect #229: XPlaneConnect plugin error in Ubuntu 18.04

surabhig11, 2020-08-06:
XplaneConnect plugin does not work with Ubuntu 18.04. I found this thread to fix it: https://github.com/nasa/XPlaneConnect/issues/151 which suggests this error is caused by ABI incompatibility between the binary we distribute and your OS. Unfortunately there's not really a good fix besides recompiling the plugin yourself for your particular platform.
However, after multiple attempts to resolve it I figured out that in XplaneConnect/xpxPlugin/CMakeLists.txt, one library (CameraCallbacks.cpp) is missing in line 18 which creates plugin load error. Compiling after this change will create the right plugin file.

P.S. This error was noticed in latest XplaneConnect release Version 1.3 RC6.

oscarkfpang, 2020-11-01:
I am using Ubuntu 18.04 and the latest XPlane-11 and I can verify that the above works for me. 

Edit the CMakeLists.txt and add CameraCallbacks.cpp as follows:

add_library(xpc64 SHARED XPCPlugin.cpp
	CameraCallbacks.cpp
	DataManager.cpp
	Drawing.cpp
	Log.cpp
	Message.cpp
	MessageHandlers.cpp
	Timer.cpp
	UDPSocket.cpp)

and compile using cmake and make. 

In case having error like "fatal error: bits/c++config.h: No such file or directory " while compiling, run this before compile: 
sudo apt-get install gcc-multilib g++-multilib

Put the compiled plugin into the XP11 plugin folder.

schmrlng, 2022-02-14:
I think this was fixed in #244; should there be a new release (i.e., 1.3 RC7) incorporating this change so that we can save some folks the trouble of compiling from source?

laelliott, 2022-02-25:
@schmrlng There probably should be a new release since it's not immediately obvious that you have to build from source. It took me a few minutes today to find the log that detail the error and then to find the threads indicating it had been solved in master. If you don't want to push to master at least adding to the readme would be helpful