nasa/XPlaneConnect #231: getDREFs returns random values

chinicuin, 2020-09-01:
Hello, I trying to read de dataref **sim/time/timer_elapsed_time_sec** (I started the timer in the simulation), and got this results:
using java 8, sometimes get java.lang.NegativeArraySizeException and others the value 1.8380204E25.
using java 11, while the timer is paused I always got 1.838034E25, and when the timer is running I get values similar to the java 8 example.
What I want to know is if I am reading the response in a correct way? Maybe that's the problem.
Here is my code:
`try(XPlaneConnect xpc = new XPlaneConnect("sim/time/timer_elapsed_time_sec",49010,49000, 4000)){
     int counter = 0;
            while(counter < 10){
                try{
                    System.out.println("--- step " + counter + " ---");
                    String[] drefs = {"sim/time/timer_elapsed_time_sec"};
                    float[][] dos = xpc.getDREFs(drefs);
                    System.out.println("value dos[0]=" + dos[0][0]);
               }catch(java.lang.NegativeArraySizeException ex){
                    System.out.println(counter + ". NegativeArraySizeException.");
                    ex.printStackTrace();
                }catch(NullPointerException ex){
                    System.out.println(counter + ". null pointer exception.");
                }
                
                Thread.sleep(10000);  // 10 seconds
                counter++;
            }
           System.out.println("Example complete");
        }
        catch (SocketException ex)
        {
            System.out.println("Unable to set up the connection. (Error message was '" + ex.getMessage() + "'.)");
        }
        catch (IOException ex)
        {
            System.out.println("Something went wrong with one of the commands. (Error message was '" + ex.getMessage() + "'.)");
        }
        catch (InterruptedException ex) {
            System.out.println("InterruptedException.");
        }
        System.out.println("Exiting");`

Originally I used **float[] dos = xpc.getDREF("sim/time/timer_elapsed_time_sec");**, but I found in this issue https://github.com/nasa/XPlaneConnect/issues/123, the problem is solved calling to DREFs instead of DREF, but for me the results are the same in both cases.

Runnong X-plane 11.41r1 in MacOS 10.14.6
Using plugin XplaneConnect 1.3rc6 (also tested with version 1.2.1)
Running the code in Windows 10
Tested In java 8 and java 11

chinicuin, 2020-09-01:
I also trying with C.
`XPCSocket sock = aopenUDP("192.168.0.5", 49010,49020);

    // Get the status of the gear handle
    char* dref = "sim/time/timer_elapsed_time_sec";
    float value = 0.0F;
    int size = 1;
    if(getDREF(sock, dref, &value, &size) < 0)
    {
        printf("And error occured.");
    }
    else
    {
        printf("The gear handle status is %f.", value);
    }

    closeUDP(sock);`

Sometimes I get this error. Response was too short. Expected at least 6 bytes, but only got 0.
Other times this error. Unexpected error size. Expected 1 rows, got -21 instead. (33, -16, -155, the number varies)

jason-watkins, 2020-09-02:
Sounds very much like an issue with packets arriving out of order or some other network issue. Unfortunately the plugin was designed around an assumption of very high network connectivity and it doesn't degrade as gracefully as I wish it did in high latency and other adverse network environments.

chinicuin, 2020-09-02:
Hi Jason.
Thanks for your reply.
I noticed something today; I was running the script and then closed x-plane in the remote machine, the thing is, the script continue to showing responses like if x-plane were running normally. I assumed I could connect to x-plane because I see in the X-plane Network configuration tab the message "Sending data to 192.168.0.8" which is the machine where I running the code. So, maybe I wasn't able to get connected at all in the first place.
In a few days, I'll give it a try in a closed subnetwork to be sure none of my coworkers are making the network slow by listening to audio streaming platforms or something. I'll let you know if I got positive or negative results.
I hope my findings can help other users.

chinicuin, 2020-09-17:
Hi. 
I got bad luck getting data with the plugin :(
So I closing this issue.
Now I'm using a javascript client https://github.com/havber/X-Plane-Datarefs-with-nodejs which doesn't need a plugin.
Thanks anyway for the support 👍

closed 2020-09-17 by chinicuin