nasa/XPlaneConnect #233: Send command to X-plane

neusat, 2020-09-29:
Can I send any command to x-plane via XPlaneConnect? Like landing lights on or off, or activate autopilot, battery on, etc.

afefer, 2020-10-06:
The plugin currently has an operation, "COMM", to momentarily trigger an X-Plane command. I've also just [submitted a pull request](https://github.com/nasa/XPlaneConnect/pull/234) to add two new operations (BCOM, ECOM) for independently beginning and ending X-Plane commands.

neusat, 2020-10-06:
Dear, thank you for your reply, I cannot find COMM manual operation on Wiki

https://github.com/nasa/XPlaneConnect/wiki/XPC-Client-Reference

How can I use it to send any command to Xplane?

https://www.siminnovations.com/xplane/command/

ICURRY-JHT, 2021-05-11:
I cannot find COMM either！！

sanderdatema, 2021-05-31:
My PR #251 was merged in the develop branch. It has the sendCOMM option.

closed 2021-06-29 by jason-watkins