nasa/XPlaneConnect #237: Weird issue with COMM functionality

MeindertN, 2020-10-24:
I'm using the new sendCOMM function to send commands to xplane. I'm using that functionality to simulate the GNS430 GPS for a project I'm working on. Overall this works great, my 3D printed GNS430 is very responsive. However, there is one particular command that only works some of the time. I'm not sure this is even an issue with XPC but I thought I'd report it here in case others have encountered this. 

I've ensured that a single command is sent when I'm expecting it to be sent. I can trace the command through my code and to the sendCOMM call.
When I look in the XPC log I consistently see the following correct entries show up: 
```
11:19:54.912|TRACE|MESG|Read message with length 27
11:19:54.912| INFO|MSGH|Handling message from 127.0.0.1:11001
11:19:54.912|DEBUG|MSGH|Existing connection. ID=0, Remote=127.0.0.1:11001
11:19:54.912|TRACE|DBUG| 43 4f 4d 4d ff 15 73 69 6d 2f 47 50 53 2f 67 34 33 30 6e 32 5f 63 6f 6d 5f 66 66
11:19:54.912|DEBUG|DBUG|Head: COMM Size: 27Type:1835627285
11:19:54.912|TRACE|COMM|Request to execute COMM command received (Conn 0)
11:19:54.912| INFO|DMAN|Executing command (value:sim/GPS/g430n2_com_ff)
11:19:54.912|DEBUG|COMM|Execute command sim/GPS/g430n2_com_ff 
```
The command only has the desired effect some of the time. I was thinking it could be a button "debouncing" issue with multiple commands being issued in short succession but through the logs I can confirm that that is not the case and that XPlane is receiving single commands.

MeindertN, 2020-10-24:
this is an issue with the xplane sdk, not XPC

closed 2020-10-24 by MeindertN

brandonharwood14, 2023-05-22:
Hi, Any chance you could share how you got the SendCOMM to work? Ive tried on my end to simulate a GNS530 but gave up after seeing that SendCOMM was forgotten about, seems a few people have ported it over but im not sure how to do that?

MeindertN, 2023-05-22:
Is your question about 

1. how to get access to the sendCOMM function. Or 
2. how to work around the original bug that I talked about

For 2. I circumvented the issue I originally posted about by using different DREFs:

To make it work, don't use 
sim/GPS/g430n2_com_ff
instead use:
sim/radios/com2_standy_flip

Don't use:
sim/GPS/g430n1_com_ff
instead use:
sim/radios/com1_standy_flip