nasa/XPlaneConnect #238: XPlaneConnect license for demo

bessszilard, 2020-10-28:
I checked the [license file](https://github.com/nasa/XPlaneConnect/blob/master/license.pdf), but it too complex for me to understand. Can I use this software to create a demo of a product? We will not redistribute because, in a commercial project, we will use another simulator, but only for a demo.

Thanks for your help.

Regards, 
Szilard

teubert, 2020-10-29:
Hello Szilard,

Thank you for your interest with XPlaneConnect. I completely understand your trouble with the license file. I personally had to go to our legal team to confirm my understanding before I felt comfortable answering your question. 

Yes you can use the software to create a demo of a product. Below is the relevant part of the license. I've been told that your activity falls under the granted rights of Reproduction and Display. I hope this helps. Let me know if you have any additional questions. ~Chris 

2.       GRANT OF RIGHTS
 
A.                  Under Non-Patent Rights:  Subject to the terms and conditions of this Agreement, each Contributor, with respect to its own contribution to the Subject Software, hereby grants to each Recipient a non-exclusive, world-wide, royalty-free license to engage in the following activities pertaining to the Subject Software:
 
1.       Use
2.       Distribution
**3.       Reproduction**
4.       Modification
5.       Redistribution
6.       Display
 
Definition from 1.J.  “Reproduction” means the making of a counterpart, image or copy of the Subject Software.

bessszilard, 2020-10-29:
@teubert thanks for the answer, that is easier to understand. 
Do we have to share our code publicly, after the demo? I mean the same way as in GPLv3.

closed 2021-06-29 by jason-watkins