nasa/XPlaneConnect #240: Resetting an X-Plane session back to initial state

conting3ncy, 2020-11-03:
**How should I reset an X-plane session back to its initial state?** 

So far I have tried sending a set of DataRefs (position, elevation, heading, etc) to X-Plane, like so:
```
        self.client.sendDREF("sim/flightmodel/position/local_vx", 0)
        self.client.sendDREF("sim/flightmodel/position/local_vy", 0)
        self.client.sendDREF("sim/flightmodel/position/local_vz", 0)

        self.client.sendDREF("sim/flightmodel/position/theta", 0)
        self.client.sendDREF("sim/flightmodel/position/phi", 0)
        self.client.sendDREF("sim/flightmodel/position/psi", 0)

        self.client.sendDREF("sim/flightmodel/position/local_ax", 0)
        self.client.sendDREF("sim/flightmodel/position/local_ay", 0)
        self.client.sendDREF("sim/flightmodel/position/local_az", 0)

        self.client.sendDREF("sim/flightmodel/position/P", 0)
        self.client.sendDREF("sim/flightmodel/position/Q", 0)
        self.client.sendDREF("sim/flightmodel/position/R", 0)

        self.client.sendDREF("sim/flightmodel/position/true_theta", 0)
        self.client.sendDREF("sim/flightmodel/position/true_phi", 0)
        self.client.sendDREF("sim/flightmodel/position/true_psi", 0) 
```
But this does not always result in a perfect reset, occasionally there are subtle changes between resets.

Is it possible to reset by reloading an X-Plane situation file? 

Additionally, could someone clarify how the reloadScript.lua is called? This does not appear to be documented beyond simply placing it in the X-Plane scripts folder.

JDatPNW, 2021-01-19:
I would like to second this issue, and I would also like to ask if it would be possible to reset the plane with a constant velocity.
When setting one of the three local/v* variables, the plane get shot up, as if the force would be applied at once, instead of the plane being steadily flying with that given speed.
Thanks

epaton2, 2021-02-16:
If you want to reset the aircraft, I would use the function loadSITU.   Or, you can use the method sendCOM and set the command with _sim/operation/reset_flight_

JDatPNW, 2021-02-16:
Could you please link me somewhere where I can look at how loadSITU works? I searched this repo, and the only mention of loadSITU is the comment you just wrote.
Thanks!

epaton2, 2021-02-16:
Hi JD,

Sorry I did not realize situation file loading is not available.  I added
it to my version.  It would be easier for you to just use the command and
restart the flight.

JDatPNW, 2021-02-16:
Hi epaton2,
could you maybe tell me how you implemented situation loading?
Thanks!

closed 2021-06-29 by jason-watkins