nasa/XPlaneConnect #247: Print text to VR screen

krao-MIT, 2021-01-29:
Hi, 

I note the sendTEXT capability, which allows printing of text to the xplane window. But I can't seem to get the text to print into the user VR view. Is there anything specific that needs to be done to enable display in VR?

Background: We're using Windows10, Matlab sendTEXT.m, HTC Vive Pro Eye headset, latest XP11 (we tried OpenGL and Vulkan rendering)

Thanks

jason-watkins, 2021-01-29:
Unfortunately the way graphics are handled in X-Plane 11 (and particularly in VR mode) seems to have changed significantly since the sendTEXT code was written. X-Plane Connect was first written for X-Plane 8 and 9 and it's starting to show its age. Sadly, I don't have the time to devote to updating it.

krao-MIT, 2021-01-29:
Thanks, Jason. I can definitely understand that. 

Do you have any recommendations for how we might provide simple information to users in our flight scenario? For example, if they're supposed to verbally report their instrument information, but they forget, how might be program in some way to prompt them? Are there sound beeps, or other forms of alerting we might use?

closed 2021-06-29 by jason-watkins