nasa/XPlaneConnect #248: Require higher precision for Latitude / Longitude data

epaton2, 2021-02-16:
Hello Jason and team,
Thank you for all the great support and resources this project has provided.  
My current goal is to manipulate the AI aircraft using the setPOSI function, but with the current XPlaneConnect, the aircraft seems jerky as it moves through the air.  This may be either a limitation in network transfer rate, X-Plane framerate,  or the lack of precision since the UDP data is transmitted as a float data type.  My hunch is that the problem is originating from the latter.    I have read other forums from users who strongly recommend using double data type for latitude/ longitude for moving the XPlane aircraft.   I also read some articles that conclude a single-precision 32bit float can only provide precision to about 1 meter in the continental U.S.A.  I discovered the current XPlaneConnect code sends and receives using the 32bit float.    

Do you see any issues with increasing the buffer size to allow 8 bytes per character instead of the current 4?  Does it make sense to do this for all data, or limit this to just the readPOSI (or getDREFs) and sendPOSI functions?   Would doing this only make matters worse because of a possible reduction in transfer rate?  My current setup uses a single computer with a client (the XPlaneConnect code compiled as a DLL) and a server (the xpcConnect plugin).   I am using the C source code and marshaling this to a C# application.

Your assistance would be greatly appreciated.
Eric Paton

epaton2, 2021-02-16:
Looks like this feature was added already. I was looking at an older version of the code. Sorry. The moderator can delete this.

closed 2021-02-16 by epaton2

donfung, 2021-03-02:
Could you please point me to where you explicitly see that doubles are accepted? I am still under the impression that the plugin only allows single type values for lat lon.