nasa/XPlaneConnect #249: MonitorExample.m unable to connect to XPlane 11 [Windows 10]

donfung, 2021-03-01:
When I run MonitorExample.m In MATLAB, I'm getting the error "java.io.IOException: No response received.".
[image]

I get the same error when trying to run Example.m as well.

I do have the XPlaneConnect plugin enabled in the Plugin Admin: 
[image]

I have tried several different port numbers but none work -- all produce the same error message.
[image]

Here are the relevant settings in the "Network" tab in XP11:
[image]
[image]

Here are the "Network Configuration" settings in the "Data Output" tab:
[image]

I am using Windows 10.  

Thanks in advanced.

donfung, 2021-03-02:
Solved. The port definition in xpcPlugin.cpp (line 77) did not match the port number in the Network settings. Changing line 77 from 49009 to `#define RECVPORT 49000` solved the issue.

closed 2021-03-02 by donfung

steagnas, 2021-11-02:
Hi donfung. 
Same problem for me.
I did not get how did you solve. I am not able to access _xpcPlugin.cpp_ . Could you please tell me how did you access it, in order to modify liine 77?
Thx

donfung, 2021-11-02:
Can you please show a screenshot of the errors you're getting? I must be honest it's been awhile since I last touched XPC

steagnas, 2021-11-03:
[image]
 That is. 

I have also noticed that on openUDP.m on line 27 you can modify a port number, but I honestly don't know which is and why.
[image]

Thanks!