nasa/XPlaneConnect #254: Missing define setCOMM in xpc.py for Python3

repeat83, 2021-07-14:
Missing define setCOMM in xpc.py for Python3 but defined for Python

ericjhkim, 2021-08-25:
I did hack together this function from the existing code:

``'
def setCOMM(self,cmd):
		buffer = struct.pack(b"<4sx", b"CMND")

		# Preconditions
		if len(cmd) == 0 or len(cmd) > 255:
			raise ValueError("CMD must be a non-empty string less than 256 characters.")

		# Pack message
		fmt = "<B{0:d}s".format(len(cmd))
		buffer += struct.pack(fmt.encode(), len(cmd), cmd.encode())

		# Send
		self.sendUDP(buffer)
'''

I pass XPlane command references ie. "sim/operations/quit" to this function, but it actually doesn't seem to do anything.
Perhaps someone else may have more luck.

sanderdatema, 2021-09-17:
Duplicate of #251. I already submitted a PR #219 for that and it was merged in the develop branch.

@jason-watkins Could you please merge dev into master? This question (understandably) keeps popping up.