nasa/XPlaneConnect #258: getDATA() function keeps timing out

ericjhkim, 2021-08-12:
I can send data to X-Plane 11 using the plugin's functions, but everytime I try calling the getDATA() function, I get an error message saying that the UDP timed out.

My code (Python 3, Windows 10):
`import xpc
import numpy as np

client = xpc.XPlaneConnect() # UDP connector
client.readDATA()`

Error message:
[image]

Perhaps I'm not using the function correctly? Strangely, all the other functions seem to work such as getDREF().

jason-watkins, 2021-08-13:
Unfortunately network issues are basically impossible to diagnose remotely. No one has ever been able to convincingly suggest that the XPC plugin is at fault, but it is odd that the other functions work. Posting your XPCLog.txt here so I can examine it would be the first step in trying to figure out what's going on.

ericjhkim, 2021-08-13:
For sure. This is the log file. 
[XPCLog2.txt](https://github.com/nasa/XPlaneConnect/files/6979280/XPCLog2.txt)

I've also attached a couple more images of the in-game settings:
[image]
[image]

From the log file, you can see I'm sending three lines of data successfully, but when I go to receive data, I get a warning saying that there are too many values when setting DREF, but I'm not quite sure what that is referring to.

ericjhkim, 2021-08-13:
Interestingly, running just getDATA() on MATLAB tells me:

[image]

MATLAB code:
`
%% Import XPC
addpath('../')
import XPlaneConnect.*
%% Setup
% Create variables and open connection to X-Plane

disp('xplaneconnect Example Script-');
disp('Setting up Simulation');
Socket = openUDP('127.0.0.1', 49009);

readDATA();
`

jason-watkins, 2021-08-13:
Okay, I have to admit that I'm confused here. The DATA functions have always been a source of issues, and we've never really been able to make them work well, but as I'm looking at the client code now, I don't understand how it's supposed to work at all (yes, I wrote the client code, but it's been years since I've looked at this particular code closely). It seems like it just blindly reads expecting something to be there, but I don't see anything in the plugin that would be constantly sending data, and even if it was it would break pretty much every other function in the client.

ericjhkim, 2021-08-13:
Ah alright, unfortunately, I'm not too well versed in network theory. Regardless, the getDREF function seems to work fine, so I'm thinking I can probably just use that to collect the data I need? I'm not quite sure what the difference is between general data and dataref, but I think all the general data values can be accessed through getDREF.

jason-watkins, 2021-08-13:
Yeah, my standard advice when people run into issues with DATA is that the DREF functions can do 99.9% of the work that DATA can.

closed 2021-08-13 by jason-watkins

ericjhkim, 2021-08-13:
Thanks for the help. I do appreciate your time and this plugin.