nasa/XPlaneConnect #259: Unstable UDP connection causing periodic timeouts

ericjhkim, 2021-08-25:
I found that if I keep a Python script connected to XPlane via UDP for too long (~20 min) the socket will timeout.
Throughout this time, I'm constantly sending and receiving DATAREFs, and the program usually times out when I'm fetching a DATAREF.

Perhaps others are experiencing something similar? I'm not sure what could be causing the socket to randomly time out in the middle of the script, since it happens at indeterminate times. It's almost like the connection is unstable.

erjdriver, 2021-11-25:
Yes I'm having the same issue - also happens if I increase the timeout to 1s and being that it's on the same machine - shouldn't really be a networking issue.

It could be xplane not responding once in a while.

Also, I think one needs to call getDREFS() not too often - because the underlying dref value doesn't change that often.

I just intercepted the socket timeout exception and ignored it and next call seems to work fine.

Any better solution?

I'm also using Python3.

zodiac1214, 2021-12-13:
why not using the native UDP from xplane?

erjdriver, 2021-12-14:
I'm not familiar with the XPlane sdk and this seems to work for now. Where can I find docs on it.

I hope this plugin is not using udp to xplane also -  because the 2x udp operation would be very slow.

zodiac1214, 2021-12-14:
there is a doc called "Exchanging Data with X-Plane" in your xplane folder and it has detailed instructions on how to work with navite UDP. it is in the instruction folder. I have used it with nodejs in two projects and it is working very well.

erjdriver, 2021-12-14:
Thanks - I'll certainly look into it.

With this plug-in, I have to poll to track datarefs I'm interested in. Do you know offhand whether the native Xplane api allows for callback notifications (i.e. no polling)?

Thanks again.

zodiac1214, 2021-12-14:
so you basically register what dataref you want, with freq, then xplane send to you over UDP. you don't have to poll at all :)

zodiac1214, 2021-12-14:
i can only speak of my apps, so i have "backend" that just gets UDP data from xplane from 5hz to 30hz depends on the phase of flight. then i only "push/callback" my frontend when a certain value changes. I don't know if that will work for you. I think it is smart for xplane to just send data out without putting too much logic. I am thinking it as an IoT device where you send data/metrics out asap and let other application to run business logic. If we let xplane doing too much more than just being a flight simulator, we are going to get lower fps :) 

I think their UDP implementation is very efficent, something like copy dataref from memory address 0x1 to 0xA and then just flush data to UDP from the new memory address. which is adding little to zero load on xplane side. If they need to maintain a callback, say something inside xplane, i think it will add a more load by just calling a function (context switching/heap/stack ....)

erjdriver, 2021-12-14:
Sure - makes sense.

Do you know whether this plugin uses UDP to communicate with xplane or something like a DLL.

If UDP, I would think it's much faster bypassing the plugin.

Would you by any chance have some sample code in Python that requests a direct dataref request.

zodiac1214, 2021-12-16:
https://github.com/charlylima/XPlaneUDP/tree/master

Check this out, you don't need anything installed into xplane. As long as you are in the same network, it will connect to xplane automatically

erjdriver, 2021-12-17:
Thank you very much - will definitely look into it.