nasa/XPlaneConnect #263: MacBook Pro with M1 Pro CPU Compilation Error

gy2256, 2021-11-01:
After git clone the repository, I went to `XPlaneConnect/xpcPlugin` and tried to build the plugin using

```
mkdir build 
cd build
cmake .. 
cmake --build .
```

and I got the following error
```
[ 15%] Building CXX object CMakeFiles/xpc64.dir/XPCPlugin.cpp.o
[ 15%] Building CXX object CMakeFiles/xpc64.dir/CameraCallbacks.cpp.o
[ 20%] Building CXX object CMakeFiles/xpc32.dir/CameraCallbacks.cpp.o
[ 15%] Building CXX object CMakeFiles/xpc64.dir/DataManager.cpp.o
[ 25%] Building CXX object CMakeFiles/xpc32.dir/XPCPlugin.cpp.o
[ 30%] Building CXX object CMakeFiles/xpc32.dir/DataManager.cpp.o
[ 35%] Building CXX object CMakeFiles/xpc32.dir/Drawing.cpp.o
[ 40%] Building CXX object CMakeFiles/xpc64.dir/Drawing.cpp.o
/Users/du/XPlaneConnect/xpcPlugin/Drawing.cpp:97:3: warning: 'glBegin' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated. (Define GL_SILENCE_DEPRECATION to silence these warnings) [-Wdeprecated-declarations]
                glBegin(GL_QUAD_STRIP);
                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl.h:2385:13: note: 'glBegin' has been explicitly marked deprecated here
extern void glBegin (GLenum mode) OPENGL_DEPRECATED(10.0, 10.14);
            ^
/Users/du/XPlaneConnect/xpcPlugin/Drawing.cpp:99:3: warning: 'glVertex3f' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated. (Define GL_SILENCE_DEPRECATION to silence these warnings) [-Wdeprecated-declarations]
                glVertex3f(x - h, y + h, z - h);
                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl.h:2740:13: note: 'glVertex3f' has been explicitly marked deprecated here
extern void glVertex3f (GLfloat x, GLfloat y, GLfloat z) OPENGL_DEPRECATED(10.0, 10.14);
            ^
/Users/du/XPlaneConnect/xpcPlugin/Drawing.cpp:97:3: warning: 'glBegin' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated. (Define GL_SILENCE_DEPRECATION to silence these warnings) [-Wdeprecated-declarations]
                glBegin(GL_QUAD_STRIP);
                ^
/Users/du/XPlaneConnect/xpcPlugin/Drawing.cpp:100:3: warning: 'glVertex3f' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated. (Define GL_SILENCE_DEPRECATION to silence these warnings) [-Wdeprecated-declarations]
                glVertex3f(x + h, y + h, z - h);
                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl.h:2740:13: note: 'glVertex3f' has been explicitly marked deprecated here
extern void glVertex3f (GLfloat x, GLfloat y, GLfloat z) OPENGL_DEPRECATED(10.0, 10.14);
            ^
/Users/du/XPlaneConnect/xpcPlugin/Drawing.cpp:101:3: warning: 'glVertex3f' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated. (Define GL_SILENCE_DEPRECATION to silence these warnings) [-Wdeprecated-declarations]
                glVertex3f(x - h, y + h, z + h);
                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl.h:2740:13: note: 'glVertex3f' has been explicitly marked deprecated here
extern void glVertex3f (GLfloat x, GLfloat y, GLfloat z) OPENGL_DEPRECATED(10.0, 10.14);
            ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl.h:2385:13: note: 'glBegin' has been explicitly marked deprecated here
extern void glBegin (GLenum mode) OPENGL_DEPRECATED(10.0, 10.14);
            ^
/Users/du/XPlaneConnect/xpcPlugin/Drawing.cpp:102:3:/Users/du/XPlaneConnect/xpcPlugin/Drawing.cpp :99warning: 'glVertex3f' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated. (Define GL_SILENCE_DEPRECATION to silence these warnings) [-Wdeprecated-declarations]:
3:                 glVertex3f(x + h, y + h, z + h);
warning:                 ^
'glVertex3f' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated. (Define GL_SILENCE_DEPRECATION to silence these warnings) [-Wdeprecated-declarations]
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl.h:2740:13: note: 'glVertex3f' has been explicitly marked deprecated here
extern void glVertex3f (GLfloat x, GLfloat y, GLfloat z) OPENGL_DEPRECATED(10.0, 10.14);
            ^
                glVertex3f(x - h, y + h, z - h);
                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl.h:2740:13: note: 'glVertex3f' has been explicitly marked deprecated here
extern void glVertex3f (GLfloat x, GLfloat y, GLfloat z) OPENGL_DEPRECATED(10.0, 10.14);
            ^
/Users/du/XPlaneConnect/xpcPlugin/Drawing.cpp:100:3: warning: 'glVertex3f' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated. (Define GL_SILENCE_DEPRECATION to silence these warnings) [-Wdeprecated-declarations]
                glVertex3f(x + h, y + h, z - h);
                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl.h:2740:13: note: 'glVertex3f' has been explicitly marked deprecated here
extern void glVertex3f (GLfloat x, GLfloat y, GLfloat z) OPENGL_DEPRECATED(10.0, 10.14);
            ^/Users/du/XPlaneConnect/xpcPlugin/Drawing.cpp:104:3: warning: 'glVertex3f' is deprecated: first deprecated in macOS 10.14 - OpenGL API deprecated. (Define GL_SILENCE_DEPRECATION to silence these warnings) [-Wdeprecated-declarations]
                glVertex3f(x - h, y - h, z + h);
                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gl.h:2740:13: note: 'glVertex3f' has been explicitly marked deprecated here
[... 363 lines cut ...]
                            "%s",
/Users/du/XPlaneConnect/xpcPlugin/Log.cpp:88:20: warning: format string is not a string literal (potentially insecure) [-Wformat-security]
                std::fprintf(fd, str);
                                 ^~~
/Users/du/XPlaneConnect/xpcPlugin/Log.cpp:88:20: note: treat the string as an argument to avoid this
                std::fprintf(fd, str);
                                 ^
                                 "%s",
2 warnings generated.
2 warnings generated.
[ 85%] Building CXX object CMakeFiles/xpc32.dir/Timer.cpp.o
[ 90%] Building CXX object CMakeFiles/xpc32.dir/UDPSocket.cpp.o
[ 95%] Linking CXX shared library XPlaneConnect/64/lin.xpl
ld: invalid option to -undefined [ warning | error | suppress | dynamic_lookup ]
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [XPlaneConnect/64/lin.xpl] Error 1
make[1]: *** [CMakeFiles/xpc64.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Linking CXX shared library XPlaneConnect/lin.xpl
ld: invalid option to -undefined [ warning | error | suppress | dynamic_lookup ]
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [XPlaneConnect/lin.xpl] Error 1
make[1]: *** [CMakeFiles/xpc32.dir/all] Error 2
make: *** [all] Error 2
```

litcoderr, 2023-08-28:
I was able to rebuild it by downloading the latest sdk and linking to it. But was not available from the plugin manager. Have you solved this problem yet?

litcoderr, 2023-08-28:
[Update]
TLDR: I've got XPlaneConnect to work on following environment setting.

Environment:
- cpu: Apple M1 Pro
- OSX version: 13.3.1

There are the steps that I took.

1. Download the latest XPlane SDK
2. Build XPlaneConnect
    a. Open in XCode
    b. Delete previous links to `XPLM` and `XPWidgets`
    c. Link downloaded `XPLM` and `XPWidgets` library file
    d. Build. There will be some unrecognized symbols. All of them were unnecessary for the moment so I just deleted them for compilation
3. Replace existing `[XPlane directory]/Resources/plugins/XPLM.framework` and `[XPlane directory]/Resources/plugins/XPWidgets.framework` with newly downloaded sdk.

and this done the job for me. Thanks.

gy2256, 2023-08-28:
Thanks for the update. I'll give it a try.

gy2256, 2023-08-31:
I can confirm this method works! Thanks for sharing it.

closed 2023-08-31 by gy2256