nasa/XPlaneConnect #264: General info question

erjdriver, 2021-11-23:
This is a general info question - i wasn't able to find a forum. Pre-apologies and I'll delete this as soon as I get pointed in the right direction.

I'm hoping I can get some help from users.

I'm writing an interface between my hardware and xplane - in python3.

I need to display some datarefs in my hardware. Is there a way to tell the plugin, update me as soon as the value changes or do I have to keep polling.

Of course polling is not efficient - what would I need to do to add that feature.

Would appreciate some sample code that does such (get notified of dataref changes) and also sets some commands.

jason-watkins, 2021-11-24:
There isn't a push-based notification system in the X-Plane Plugin SDK API that XPC exposes, so you would have to petition the X-Plane developers to add such functionality to the plugin SDK before such a feature could be fully implemented.

The network architecture the plugin uses also isn't really suited to sending push notifications and I suspect it would have to be completely re-written. This is something I've wanted to do for a long time, but I'm no longer paid to work on XPC and I haven't been able to find enough free time to do it for fun yet.

erjdriver, 2021-11-24:
Thank you for the response. Yes polling is not a good way to go but looks like no options.

Actually creating a callback notification mechanism in the plugin is doable - but really no more efficient because the plugin will have to do the polling.

The solution has to come from xplane - and looks like it's not in the next version of sdk either.

It's a little bit more work for them - but it'll prevent slowdown of the sim when users just add plugins without thinking about the ramifications.

Anyway thanks - poll I will.

closed 2021-11-24 by erjdriver