nasa/XPlaneConnect #277: How do we send array datarefs? For example, "sim/flightmodel/engine/ENGN_thro"

GBry, 2022-03-27:
I have tried :
- sending the dref as a list (8 items)
- converting the list into string before sending
- send DREfs and putitng [i] behind, like "sim/flightmodel/engine/ENGN_thro[0]"

None of this work. Is this not implemented yet?

jason-watkins, 2022-03-27:
[The wiki](https://github.com/nasa/XPlaneConnect/wiki/sendDREFs) has documentation and examples. It would be helpful to know what client language you are using.

GBry, 2022-03-27:
I.. Already edited the title and added the language within seconds of
posting.. But OK, I'm using Python 3, sorry for the confusion.

GBry, 2022-04-03:
@jason-watkins in the examples it only shows how to send multiple DREFs with an array, it doesn't show us how to send a DREF with an array input, hence why I'm asking.

iftahnaf, 2022-12-06:
Hi all,
Any solution to the same exact question?

iftahnaf, 2022-12-07:
Thanks all, this is working now using your help and the fix in #242.

AlvaroSierra, 2022-12-07:
> > For any value of an array pass the DREF with an additional "[i]" for index i at the end. For example `sim/multiplayer/controls/yoke_pitch_ratio[0]` for the zero value of that array.
> > This is what is used for plugins so I assume it would be the same for XPlane Connect.
> 
> Ah, so you have to actually iterate each value?

Yes, you will have to send each value separately.

AlvaroSierra, 2022-12-08:
@jason-watkins you might want to close this issue.

closed 2022-12-08 by jason-watkins

IanOMHiggins, 2023-08-03:
I have this exact problem, but this solution is not working. Using python3 and BETA eVTOL.

After doing:
client.sendDREF('sim/operation/override/override_throttles',True)

client.sendDREF('sim/flightmodel/engine/ENGN_thro_use',1.)
#commands max thrust to only the front right propeller, so I know this dref is working, but

client.sendDREF('sim/flightmodel/engine/ENGN_thro_use[0]',1.)
client.sendDREF('sim/flightmodel/engine/ENGN_thro_use[1]',1.)
client.sendDREF('sim/flightmodel/engine/ENGN_thro_use[2]',1.)
...
#does nothing.

I've also tried the 'sim/flightmodel/engine/ENGN_thro' DREF, but this does nothing in either case.
Also, OP explicitly said he tried this and it didn't work, so I'm surprised @iftahnaf got it to work with this. Doesn't seem to work for me though...
Am I missing something?

IanOMHiggins, 2023-08-04:
I was actually able to make this work by adding the change in #242. (line 313 in python3's xpc.py should be *value instead of value for the final argument of struct.pack). It says this change is merged, but it doesn't seem to be in the master branch. I still couldn't get the 'sim/flightmodel/engine/ENGN_thro_use[0]' to work, but I can at least now use sendDREF to send arrays. One disadvantage of this approach is I cannot write to engine 3 without also writing to engines 0, 1, and 2, etc., but I can just read off of the DREF for the current engine throttle and pass those values in.

eduardoimanuel, 2023-11-05:
> I was actually able to make this work by adding the change in #242. (line 313 in python3's xpc.py should be *value instead of value for the final argument of struct.pack). It says this change is [...]

@IanOMHiggins I am having same issue here. For my project, I only need to send different signals for each engine so I can compromise with that disadvantage. I also write `sim/operation/override/override_throttles` value to True so my Python script are able to sent the throttle signal to each of my engines.