nasa/XPlaneConnect #287: XPlaneConnect is not working with XPlane 11

pd3a, 2022-09-20:
I have followed the procedure mentioned in Quick Start. But i cannot see XPlaneConnect in XPlane 11 plugins. I'm new to this.

qyzoom, 2022-11-24:
Me too, how to solve this problem. I am very anxious.Have you solved it, if so, please tell me how to deal with it.

qyzoom, 2022-11-24:
I have known how to solve this problem, you should install XPlaneConnect-1.3-rc1 rather than newer versions.

9nol, 2022-12-21:
Hello, I face the same problem  here. I have tried the different Xplaneconnect versions but none would appear into the XP App plug-in list. I run Xp11.55 on Mac 12.6.2 Monterey. Thanks for any help.
[xfse_log.txt](https://github.com/nasa/XPlaneConnect/files/10280072/xfse_log.txt)
[XPCLog.txt](https://github.com/nasa/XPlaneConnect/files/10280073/XPCLog.txt)