nasa/XPlaneConnect #296: Connection reset error

MarsLin7, 2023-03-15:
Hi,

Thank you so much for the amazing work.

When I try to run the monitorExample.py there's a connection reset error in file xpc.py, both line 69 - "return self.socket.recv(16384)" and line 159 - "resultBuf = self.readUDP()". The error message is as follows:
ConnectionResetError: [WinError 10054] An existing connection was forcibly closed by the remote host.

I did successfully run the example file with no errors before and now, with no modification to the code, I'm getting this error somehow. I'm new to coding so any solutions or suggestions are much appreciated.

AlvaroSierra, 2023-03-15:
This means that it cannot connect to the plugin either because it is not running or because a firewall is restricting it.

Check on X-Plane if XPlaneConnect is running by looking into the Plugins drop down to check if it appears.

Notenlish, 2024-07-19:
I'm having the same problem, XPlaneConnect is working and I've also tried disabling network protection, still it doesn't work. Does anyone have any solutions?