nasa/XPlaneConnect #308: Command horizontal and vertical throttles

IanOMHiggins, 2023-08-03:
Hello,

I wish to fly the BETA eVTOL while manually control the vertical throttle with python3. This can be done with a joystick, but I see no writable dataref for this. I could in theory control each propeller (although I have had problems with this; commented on another issue) with a quadrotor dynamics controller, but I would ideally like to only alter the "Throttle Vertical" just like the joystick can do. Is this possible?

Thanks
Ian

YovanDharwal, 2023-08-05:
Hi @IanOMHiggins   you can use ''sim/cockpit2/engine/actuators/throttle_ratio_all '' to send throttle command one by one. You can define number of motors  throttleRatios[0] = 0.5; // 
// 			throttleRatios[1] = 0.5; 
// 			throttleRatios[2] = 0.1; 
// 			throttleRatios[3] = 0.1; 
0 for 1st motor and so on. Are you able to give throttle to your VTOL motors manually using joystick?? I have one Quad copter where i am able to control all the motors manually and by sending commands from C code as well, but a different VTOL hybrid bird am not able to give throttle manually to VTOL motors and neither from my code (total 5 motors Quad + puller). If you can share the settings with me for Quad motors or the ".acf" it will be really helpfull.

IanOMHiggins, 2023-08-09:
Hi, when I said 'manually', I actually meant 'through code' (I am using joystick input to inform my python api calls, bypassing xplane's joystick reader).

Your answer is exactly what I was looking for! interestingly, the 'sim/cockpit2/engine/actuators/throttle_ratio_all' DREF has four separate values for "vertical throttle", but they are always identical, even when the four engines are different speeds (due to roll/pitch commands, etc). Setting all four at once to the same value still allows the stick commands to provide roll/pitch moments, which is the exact behavior I desire. Thanks very much!

YovanDharwal, 2023-08-14:
if values are identical that means its just for lift throttle(if you know the flight dynamics you will understand)