nasa/XPlaneConnect #309: Please adjust for SDK release 4 so that we can use with X-Plane 12

devleaks, 2023-09-24:
Simple recompilation does not work, some defines missing, some functions changed.
Please add MacOS arm64 to binary distribution list.
Thank you

gameengineer, 2023-12-22:
Ditto. I tried building this plugin project using latest X-Plane 12 SDK 4.0.1 and some deprecated objects were being used relating to Drawing waypoints. Interestingly enough there were only 3 errors total.

pregatori, 2024-09-11:
I did build this project using the latest X-Plane 12 SDK 4.1.0, for this I simply ignored the deprecated objects. 
With this I could generate a mac.xpl executable compatible with arm64 macOS binary, making XPlane12 a lot faster. 

Check my forked repo to download the code: 
https://github.com/pregatori/XPlaneConnect/tree/XPCPlugin-XPlane-4.1.0-SDK

B00TK1D, 2024-09-17:
@pregatori Would you by chance be willing to post your compiled mac.xpl?  I tried cloning your fork and building, but xcode refuses to build (I'm sure due to my own environment issues, but it would be helpful to have the release available somewhere for others as well I'm sure).  Easiest way would be just to make a release on your fork, or you can attach it as a file here.